ros2/geometry2 #707: [Jazzy] AttributeError: 'PoseStamped' object has no attribute 'position'

jkaniuka, 2024-09-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - Debian packages inside Docker container
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1. Launch static transform publisher
`ros2 run tf2_ros static_transform_publisher 5.0 10.0 15.0 0.0 0.0 0.0 1.0 odom base_link`

2. Launch the following node.
```python
import tf2_geometry_msgs
import rclpy
from rclpy.node import Node
from geometry_msgs.msg import Pose, PoseStamped
import tf2_ros

class TFListener(Node):
    def __init__(self):
        super().__init__("tf_listener_node")

        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self)

        # Timer to periodically check for the transform
        self.timer = self.create_timer(1.0, self.on_timer)
        self.base_link_to_odom = None

    def on_timer(self):
        try:
            self.base_link_to_odom = self.tf_buffer.lookup_transform(
                "odom", "base_link", rclpy.time.Time()
            )
            self.get_logger().info("Transform between odom and base_link is available.")

            pose_in_base_link = Pose()
            pose_in_base_link.position.x = 1.0
            pose_in_base_link.position.y = 2.0
            pose_in_base_link.position.z = 0.0
            pose_in_base_link.orientation.x = 0.0
            pose_in_base_link.orientation.y = 0.0
            pose_in_base_link.orientation.z = 0.0
            pose_in_base_link.orientation.w = 1.0

            pose_stamped = PoseStamped()
            pose_stamped.pose = pose_in_base_link
            pose_stamped.header.stamp = self.get_clock().now().to_msg()
            pose_stamped.header.frame_id = "base_link"

            # Transform the pose from base_link to odom
            transformed_pose_stamped = tf2_geometry_msgs.do_transform_pose(
                pose_stamped, self.base_link_to_odom
            )

        except Exception as e:
            self.get_logger().warn(f"Could not get transform: {e}")

def main(args=None):
    rclpy.init(args=args)
    node = TFListener()
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == "__main__":
    main()

```

#### Expected behavior
I would expect that the pose_stamped is transformed from one frame to another

#### Actual behavior
`AttributeError: 'PoseStamped' object has no attribute 'position'`

#### Additional information

```
  File "/opt/ros/jazzy/lib/python3.12/site-packages/tf2_geometry_msgs/tf2_geometry_msgs.py", line 352, in do_transform_pose
    pose.position.x,
    ^^^^^^^^^^^^^
AttributeError: 'PoseStamped' object has no attribute 'position'
```

ahcorde, 2024-09-02:
You should use:

```python
            transformed_pose_stamped = tf2_geometry_msgs.do_transform_pose_stamped(
```

ahcorde, 2024-09-02:
In the future, please use one of the supported channels for help requests like this, such as the [Robotics Stack Exchange](https://robotics.stackexchange.com/). GitHub tickets should be used for bugs and enhancement tracking, not technical assistance.

closed 2024-09-02 by ahcorde