ros2/geometry2 #708: tf2_tools.view_frames and tf2_ros.tf2_monitor - allow configurable topics

Ryanf55, 2024-09-11:
## Feature request

#### Feature description

In ArduPilot, we namespace all the TF topics. We want to run the view_frames tool on the frames published by ArduPilot, but we can't unless we remap them. In a large system, it's nice to be able to run diagnostics without changing it. 

#### Implementation considerations

Add the following options
* `tf_static_topic` which defaults to the exist `tf_static`
* `tf_topic` which defaults to the existing `tf`

Example usage:
`ros2 run tf2_tools view_frames --tf_static_topic=/ap/tf_static --tf_topic=/ap/tf`

clalancette, 2024-09-11:
That seems like a reasonable feature request to me.  Please feel free to open a PR implementing it, we are happy to review.

Ryanf55, 2024-09-11:
Thanks, I'll put this together. Please assign me.