ros2/geometry2 #710: time jump post_callback called with wrong number of arguments

ottojo, 2024-10-03:
## Bug report

This manifests as a crash in an rqt plugin through tf2_ros, but i have reason to believe the actual issue lies in rclpy... It seems like the time jump callback gets called incorrectly:

```
jonasotto@ade ➜  workspace rqt --ros-args -p use_sim_time:=True                                                                                                                 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-jonasotto'
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_gui_py/rclpy_spinner.py", line 32, in run
    executor.spin_once(timeout_sec=1.0)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 899, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 896, in _spin_once_impl
    future.result()  # raise any exceptions
    ^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/task.py", line 94, in result
    raise self.exception()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 508, in handler
    await call_coroutine()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 396, in _execute
    await await_or_execute(sub.callback, *msg_tuple)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 111, in await_or_execute
    return callback(*args)
           ^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/time_source.py", line 128, in clock_callback
    clock.set_ros_time_override(time_from_msg)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/clock.py", line 364, in set_ros_time_override
    self.handle.set_ros_time_override(time._time_handle)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/clock.py", line 242, in callback_shim
    original_callback(TimeJump(clock_change, duration))
TypeError: Buffer.time_jump_callback() takes 1 positional argument but 2 were given
[1]    295333 IOT instruction (core dumped)  rqt --ros-args -p use_sim_time:=True
```

The callback does actually only take 1 argument (https://github.com/ros2/geometry2/blob/jazzy/tf2_ros_py/tf2_ros/buffer.py#L107), but at least according to the type hints that should be correct (https://github.com/ros2/rclpy/blob/jazzy/rclpy/rclpy/clock.py#L221)... Not sure what goes wrong here.

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - debian packages
- Version or commit hash:
  - rclpy 7.1.2 in ros 2 jazzy
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
* run rqt with sim-time enabled (`rqt --ros-args -p use_sim_time:=True`), open tf tree visualization
* run rosbag replay with tf data and clock (`ros2 bag play --clock 100 ...`)
* restart rosbag replay
* observe crash

#### Expected behavior
Time jump callbacks called correctly, resulting action taken (tf buffer clear etc)

#### Actual behavior
Crash during execution of time jump callback

clalancette, 2024-10-03:
See #711, which should fix this.

ottojo, 2024-10-03:
> This is a bug in `tf2_ros_py`. The `time_jump_callback` method needs a `self` argument. The [jump callback in tf2_ros](https://github.com/ros2/geometry2/blob/da10f77d98ab97d2dfdfa9bfc4c98042a8ad7c3d/tf2_ros_py/tf2_ros/buffer.py#L107-L109) is a [method](https://docs.python.org/3/tutorial/classes.html#method-objects). This is not the same as the [time jump callback in clock.py](https://github.com/ros2/rclpy/blob/2de5c8ea447a1614a7c36d0fcf2250cefc21a594/rclpy/rclpy/clock.py#L250-L262). That one is a function that happens to be defined in the scope of a method.
> 
> The type annotation `def time_jump_callback(time_jump: TimeJump):` is incorrect. The first argument of a method is always bound to an instance of the class, and by convention this argument is usually called `self`. If I fix the argument name and the type annotation then the reason for the error becomes clear: `def time_jump_callback(self: "Buffer"):`. `Buffer.time_jump_callback()` takes 1 positional argument (a Buffer instance) but 2 were given (a Buffer instance and a `TimeJump` instance).

great find, i would not have found that...
interestingly python only crashes once the method is actually called, not when it is referenced as `buffer.time_jump_callback` (last line in my example). Notably, pyright throws like 4 errors, maybe something like that could be part of CI in the future, given that we already have type annotations in many places

Code sample in [pyright playground](https://pyright-play.net/?code=MYGwhgzhAEAqCmEAuBhcUBcAoL0-QBN4AzaAfTIEsA7SpCgCgnhGIEpt8vpnWA6AOYAnePCQ0B0ALzQARAAkWIAPaycXIqRrIw1YPDIBbMQAtlBJi2IAaaNTDGMPJEI65ueAA5CaSS-2FRcWoBW1liIWVDaG0kXX1oYyQzAmtZW3tjNnV8TWhQSAgjU3MGTPgnZFdOD2hvX39iQRExCTDFEBVoCKj89BgklLSMh3hsnCRpOERUfoZswfMppD4CqGLk8yxFi1kAd2UhEAJZbKA)

```python
class TestClass:

    def __init__(self):
        self.greeting = "Hello"

    def instance_method(self, name: str):
        print(self.greeting, "from instance method,", name)

    def class_method(name: str):
        print(self.greeting, "Hello from class method,", name)

t = TestClass()
method = t.class_method
method("world")

```

PR #711 (2024-10-04): Fix the time_jump_callback signature.
Because it accesses object data, the first argument must be 'self'.

This should fix #710
PR #712 (2024-10-04): Fix the time_jump_callback signature. (backport #711)
Because it accesses object data, the first argument must be 'self'.

This should fix #710 <hr>This is an automatic backport of pull request #711 done by [Mergify](https://mergify.com).
closed 2024-10-04 by ahcorde