ros2/geometry2 #717: tf2_echo: specifying quaternion order (right now ambiguous if xyzw or wxyz)

abhishek47kashyap, 2024-10-24:
## Feature request

#### Feature description

When echoing transformation between two frames using `ros2 run tf2_ros tf2_echo foo bar`, the resulting logs containing`Rotation: in Quaternion` does not specify whether the ordering of the 4 elements is `xyzw` or `wxyz`.

Peeking inside the source code, the order is `xyzw`:
https://github.com/ros2/geometry2/blob/ros2/tf2_ros/src/tf2_echo.cpp#L153-L154

I propose specifying this order in the output of `tf2_echo`.

#### Implementation considerations

A single `std::cout()` needs to be changed in `tf2_ros/src/tf2_echo.cpp` (linked above).

ahcorde, 2024-10-25:
Hi @abhishek47kashyap,

do you mind open a PR with you suggestions? Happy to review it

abhishek47kashyap, 2024-10-25:
@ahcorde Thanks for responding. PR's up!

PR #718 (2024-10-28): tf2_echo: specifies quaternion order to be xyzw
Resolves #717

Targets `rolling` but can be backported to earlier distros that haven't reached EOL (`jazzy`, `iron`, `humble`) because the change is only a single `std::cout`.
closed 2024-10-28 by ahcorde