ros2/geometry2 #721: fatal error: geometry_msgs/msg/velocity_stamped.hpp: No such file or directory

Kalemat96, 2024-10-28:
## Bug report

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  -  humble 0.25.8

#### Steps to reproduce issue
include "tf2_geometry_msgs/tf2_geometry_msgs.hpp" in a custom package and build it.

#### Expected behavior
Installation using apt works fine.

#### Actual behavior
After update the custom package gives an error while building:

/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp:51:10: fatal error: geometry_msgs/msg/velocity_stamped.hpp: No such file or directory
   51 | #include "geometry_msgs/msg/velocity_stamped.hpp"

#### Additional information
Today I updated all the ros packages and I get the error while building.

I then built the 0.25.7 version of the geometry2 package locally and it works again, but I couldn't figure out the problem.

Does anyone have an idea?

Kalemat96, 2024-10-29:
Thanks for your reply, I just updated all packages. I have installed `ros-humble-geometry-msgs` version 4.2.4.

I can find the file `VelocityStamped.msg` file in the humble folder, but somehow it is not installed properly.

It is not listed up using `ros2 interface list | grep Velocity`.

christophebedard, 2024-10-30:
I do have it on my Ubuntu 22.04 system with Humble:

```
$ source /opt/ros/humble/setup.bash
$ ros2 interface list | grep VelocityStamped
    geometry_msgs/msg/VelocityStamped
```

Try reinstalling all packages with `sudo apt-get install --reinstall ros-humble-desktop` (or `ros-humble-ros-base`).

closed 2024-11-08 by mjcarroll

AbdoullahNdoye, 2024-12-10:
Hi, did someone found a solution ? I have the same issue. reinstalling all packages didn't help and I have also geometry_msgs version >= 4.2.4

yigitboracagiran, 2024-12-13:
I solved my problem, it was related to workspaces...
Can you check your AMENT_PREFIX_PATH with `echo $AMENT_PREFIX_PATH`?

ptientho, 2025-06-29:
[image]