ros2/geometry2 #742: Add isNan or isValid to Vector and Quaternion

kyle-basis, 2024-12-05:
#### Feature description
It would be great for validation if Vector and Quaternion had validity checks built in, for NaN (and magnitude on Quaternion).

See https://github.com/ros2/geometry2/pull/741/files#r1868535923

It's easy and fast to check math type validity

#### Implementation considerations

N/A

obamwonyi, 2024-12-24:
Hi can I work on this ?

GogiPuttar, 2024-12-31:
Hi, I'd like to work on this too if it's still available. Just to clarify, should I wait for the "Remove many extra conversions..." PR to get merged or can I start working on this independently?

k3rnel-pan1c-a, 2025-03-08:
Hello, I was wondering if I could start working on this, do I just get started or what?

PR #780 (2025-04-24): Add isnan support
Closes #742
closed 2025-04-24 by ahcorde