ros2/geometry2 #743: spin_thread is not working in rclpy.

Tacha-S, 2024-12-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - binaries
- Version or commit hash:
  - 7.1.2
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

The spin thread is not working as intended.
As mentioned in #438, when running a transform listener on the same node as the main process with a single-threaded executor specified, the TF updates can be delayed. (See the example code below.)

To avoid this, you can either create a separate node for the transform listener or switch the main executor to a multi-threaded executor. While these workarounds exist, it concerns me that the spin_thread argument alone is almost meaningless.

I understand this may not be the best solution, but for example, in rclcpp, a separate node is internally created for the transform listener. Wouldn't it have been possible for rclpy to do the same?

#### Steps to reproduce issue

```
import time

import rclpy
from rclpy.node import Node
from rclpy.time import Time
import tf2_ros

class TFListener(Node):

    def __init__(self, node_name):
        super().__init__(node_name)
        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self, spin_thread=True)

        self.timer = self.create_timer(0.01, self.timer_callback)
        self.timer2 = self.create_timer(0.1, self.timer2_callback)

    def timer_callback(self):
        time.sleep(1)

    def timer2_callback(self):
        try:
            transform = self.tf_buffer.lookup_transform('base_link', 'map', Time())
            self.get_logger().info(
                f'Latest transform stamp: {transform.header.stamp.sec + transform.header.stamp.nanosec * 1e-9}')
        except Exception as e:
            self.get_logger().error(str(e))

if __name__ == '__main__':
    rclpy.init()
    node = TFListener('test_tf_listener')
    rclpy.spin(node)
    rclpy.shutdown()
```

#### Expected behavior

#### Actual behavior

#### Additional information

clalancette, 2024-12-17:
> I understand this may not be the best solution, but for example, in rclcpp, a separate node is internally created for the transform listener.

That's actually not true; the C++ code uses the same node.  However, when `spin_thread` is true it does create a different callback group for the `/tf` and `/tf_static` subscriptions, and then put those in a dedicated thread with its own executor: https://github.com/ros2/geometry2/blob/84dd504fdfe1e3e0257e07e9a05e5b73a8ba2261/tf2_ros/include/tf2_ros/transform_listener.h#L186-L216 .

The tf2_ros_py code isn't as sophisticated as this, and instead just creates a thread with an executor, and adds the node to it: https://github.com/ros2/geometry2/blob/84dd504fdfe1e3e0257e07e9a05e5b73a8ba2261/tf2_ros_py/tf2_ros/transform_listener.py#L110-L119 .  That means that if there are other things in the node, they can compete with the `/tf` and `/tf_static` topics, since they are all in the same callback group.

Probably the correct solution here is to make the tf2_ros_py code more like the tf2_ros C++ code, and add a new callback group to the executor.  However, I'll note that the rclpy executor code isn't nearly as sophisticated as the rclcpp executor code, so there is some possibility this will also require changes to rclpy.

Tacha-S, 2024-12-18:
I thought a new node was being created here, but is that not?
https://github.com/ros2/geometry2/blob/84dd504fdfe1e3e0257e07e9a05e5b73a8ba2261/tf2_ros/src/transform_listener.cpp#L46-L68

Tacha-S, 2024-12-18:
> The `tf2_ros_py` code simply creates a thread with an executor and adds the node to it.

This is correct, but it seems that if you spin the same node with another executor, the executor that was supposed to run in a separate thread stops functioning.

In the following script, the `rclpy.spin_once(node)` line effectively overwrites the executor. Just having this line causes the TF thread to stop working, while removing it allows the thread to function correctly.

```python
import rclpy
from rclpy.duration import Duration
from rclpy.node import Node
from rclpy.time import Time
import tf2_ros

class TFListener(Node):

    def __init__(self, node_name):
        super().__init__(node_name)
        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self, spin_thread=True)

    def run(self):
        while rclpy.ok():
            try:
                transform = self.tf_buffer.lookup_transform('base_link', 'map', Time())
                self.get_logger().info(
                    f'Latest transform stamp: {transform.header.stamp.sec + transform.header.stamp.nanosec * 1e-9}')
            except Exception as e:
                self.get_logger().error(str(e))
            self.get_clock().sleep_for(Duration(seconds=0.1))

if __name__ == '__main__':
    rclpy.init()
    node = TFListener('test_tf_listener')
    rclpy.spin_once(node)  # overwrite executor?
    node.run()
    rclpy.shutdown()
```

This might be because the `rclpy` executor adds nodes instead of callback groups, resulting in the node being taken over.