ros2/geometry2 #746: C++ headers were not deprecated consistently

sea-bass, 2025-01-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - rolling
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

In #720, several `.h` files were converted to `.hpp` with deprecation warnings.

However, this was not done consistently throughout the repo. For example, the `tf2_ros` package still is missing several `.hpp` files. This makes it pretty inconvenient to do blanket search/replaces in downstream repos that use this.

#### Expected behavior

Every header should consistently have a `.hpp` counterpart now.

#### Actual behavior

Not the case currently, some are still missing.

#### Additional information

N/A

ahcorde, 2025-01-06:
Related PR https://github.com/ros2/geometry2/pull/744

clalancette, 2025-01-07:
> However, this was not done consistently throughout the repo. For example, the `tf2_ros` package still is missing several `.hpp` files. This makes it pretty inconvenient to do blanket search/replaces in downstream repos that use this.

For what it is worth, we usually do it on a package-by-package basis, not for the whole repository.  It is just too disruptive to do everything at once.  With #744 now in, I think the big remaining package to do here is tf2_ros, which will require another large round of downstream PRs.

CursedRock17, 2025-01-09:
Hey sorry to be late to the conversation, I've been pretty stacked up with some stuff. Like @clalancette said, I assumed we were going to tackle this at the package level. Especially since #259 (the original target) only mentioned `tf2`. I wouldn't mind doing `tf2_ros` as well, but seeing the bit of chaos #720 has created, I would recommend waiting for this chunk to settle in. I can create an issue and have it assigned to me if interested.

sea-bass, 2025-01-09:
This makes sense!

For what it's worth, this "chaos" is a good thing. It means lots of CI pipelines that use this package are treating warnings as errors and reacting quickly to the changes.

I'm happy to close / rename this issue in favor of tracking tickets to consistently port all the other individual packages in this repo.

Though I will say, we did this for `moveit2` recently ad well, with the conscious effort of ensuring all the changes to the binaries happened in the same sync.

christophebedard, 2025-01-17:
> I'm happy to close / rename this issue in favor of tracking tickets to consistently port all the other individual packages in this repo.

That would indeed be good!

CursedRock17, 2025-01-17:
I would be in agreement to just renaming this issue since there's already discussion about the topic here. The only package we need change header files for, in this repo, is `tf2_ros`. Everything else would just be adjustments for that package.

> all the changes to the binaries happened in the same sync.

Seems to be the next biggest problem in from #720's release.

closed 2025-07-18 by ahcorde