ros2/geometry2 #756: Python API isn't available for jazzy

EmmanuelMess, 2025-02-02:
I am looking at http://docs.ros.org/en/jazzy/p/tf2_ros_py/ (linked from https://index.ros.org/p/tf2_ros_py/github-ros2-geometry2/) and I dont find the python API even though the code is documented.

christophebedard, 2025-02-03:
I think this is an issue with rosdoc2, not specifically this package. See https://github.com/ros-infrastructure/rosdoc2/issues/157.

christophebedard, 2025-02-03:
Although in this case here it was fixed in the package: https://github.com/ros2/launch_ros/pull/425. Not sure if it's the same issue.

EmmanuelMess, 2025-02-03:
There were a few issues with the same symptoms, can you check the rosdoc2 job logs?

christophebedard, 2025-02-03:
This should be it: https://build.ros2.org/view/Jdoc/job/Jdoc__geometry2__ubuntu_noble_amd64/14/consoleFull

EmmanuelMess, 2025-02-03:
The first error I see is:

```
ERROR: Invalid value `None` in intersphinx_mapping['https://docs.python.org/']. Expected a two-element tuple or list.
```

Which [here](https://github.com/jupyterlab/jupyterlab/issues/14422#issuecomment-1520083065) was fixed by changing `intersphinx_mapping = {'https://docs.python.org/': None}` to `intersphinx_mapping = {'python': ('https://docs.python.org/3', None)}`

Then it starts working on `tf2_kdl`.

christophebedard, 2025-02-04:
Nice catch! Yes, I've also seen this fix elsewhere in the past few months.

EmmanuelMess, 2025-02-09:
Okay, this should work now. There are a few minor issues like a missing "Python API" button in the bar. And for cpp there seems to be *a lot* of warnings.

closed 2025-02-09 by EmmanuelMess