ros2/geometry2 #765: Memory access error in the tf2_ros::BufferInterface::transform function

zhihaoshang, 2025-03-10:
# Bug Report
## Environment
OS Version: Ubuntu 24.04
geometry2 version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc
## Description
A Memory Access Violation (Segmentation Fault, SEGV) was detected using AddressSanitizer while unit testing the tf2_eigen library. This error occurs in the tf2_ros::BufferInterface::transform function, which involves a transform operation of type Eigen::Transform.

## Steps to reproduce

```C++
// To get M_PI, especially on Windows.
#ifdef _MSC_VER
#ifndef _USE_MATH_DEFINES
#define _USE_MATH_DEFINES
#endif
#endif
#include <cmath>
#include <memory>
// Version 3.4.0 of Eigen in Ubuntu 22.04 has a bug that causes -Wclass-memaccess warnings on
// aarch64.  Upstream Eigen has already fixed this in
// https://gitlab.com/libeigen/eigen/-/merge_requests/645 .  The Debian fix for this is in
// https://salsa.debian.org/science-team/eigen3/-/merge_requests/1 .
// However, it is not clear that that fix is going to make it into Ubuntu 22.04 before it
// freezes, so disable the warning here.
#if defined(__GNUC__) && !defined(__clang__)
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wclass-memaccess"
#endif
#include <Eigen/Geometry>  // NOLINT
#if defined(__GNUC__) && !defined(__clang__)
#pragma GCC diagnostic pop
#endif
#include "gtest/gtest.h"
#include "geometry_msgs/msg/point.hpp"
#include "geometry_msgs/msg/point_stamped.hpp"
#include "geometry_msgs/msg/pose.hpp"
#include "geometry_msgs/msg/pose_stamped.hpp"
#include "geometry_msgs/msg/transform_stamped.hpp"
#include "rclcpp/clock.hpp"
#include "tf2/convert.hpp"
#include "tf2/transform_datatypes.hpp"
#include "tf2_eigen/tf2_eigen.hpp"
#include "tf2_ros/buffer.h"
#include "tf2_ros/transform_listener.h"

TEST(TfEigen_v2, ConvertVector3dStamped)
{
  const tf2::Stamped<Eigen::Vector3d> v(Eigen::Vector3d(1, 2, 3), tf2::TimePoint(
      std::chrono::seconds(5)), "test");
  tf2::Stamped<Eigen::Vector3d> v1;
  geometry_msgs::msg::PointStamped p1;
  tf2::convert(v, p1);
  tf2::convert(p1, v1);
  EXPECT_EQ(v, v1);
}

TEST(TfEigen_v2, ConvertAffine3dStamped)
{
  const Eigen::Affine3d v_nonstamped(Eigen::Translation3d(1, 2, 3) * Eigen::AngleAxis<double>(
      1, Eigen::Vector3d::UnitX()));
  const tf2::Stamped<Eigen::Affine3d> v(v_nonstamped, tf2::TimePoint(
      std::chrono::seconds(42)), "test_frame");
  tf2::Stamped<Eigen::Affine3d> v1;
  geometry_msgs::msg::PoseStamped p1;
  tf2::convert(v, p1);
  tf2::convert(p1, v1);
[... 195 lines cut ...]
==11802==The signal is caused by a READ memory access.
==11802==Hint: address points to the zero page.
    #0 0x602af8fcd91f in tf2::Stamped<Eigen::Transform<double, 3, 2, 0> >& tf2_ros::BufferInterface::transform<tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > >(tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > const&, tf2::Stamped<Eigen::Transform<double, 3, 2, 0> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) const (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x19291f) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #1 0x602af8fc503f in tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > tf2_ros::BufferInterface::transform<tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > >(tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) const (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x18a03f) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #2 0x602af8fb73ea in void EigenBufferTransform_v2::testEigenTransform_v2<2>() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x17c3ea) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #3 0x602af8fa2519 in EigenBufferTransform_v2_Affine3d_v2_Test::TestBody() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x167519) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #4 0x602af90cc9fb in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x2919fb) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #5 0x602af90b9c99 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x27ec99) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #6 0x602af905fa07 in testing::Test::Run() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x224a07) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #7 0x602af90611e3 in testing::TestInfo::Run() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x2261e3) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #8 0x602af906253c in testing::TestSuite::Run() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x22753c) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #9 0x602af9089267 in testing::internal::UnitTestImpl::RunAllTests() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x24e267) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #10 0x602af90cfe56 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x294e56) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #11 0x602af90bcf36 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x281f36) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #12 0x602af9085871 in testing::UnitTest::Run() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x24a871) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #13 0x602af8facb71 in RUN_ALL_TESTS() (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x171b71) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #14 0x602af8fa823f in main (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x16d23f) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)
    #15 0x79c32002a1c9 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #16 0x79c32002a28a in __libc_start_main_impl ../csu/libc-start.c:360
    #17 0x602af8f9e154 in _start (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x163154) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/shangzh/workspace/templated_tests/geometry2/build/tf2_eigen/tf2_eigen-test+0x19291f) (BuildId: 8511f3c8a27d9d1aea63f032fc3dca602c41f2fa) in tf2::Stamped<Eigen::Transform<double, 3, 2, 0> >& tf2_ros::BufferInterface::transform<tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > >(tf2::Stamped<Eigen::Transform<double, 3, 2, 0> > const&, tf2::Stamped<Eigen::Transform<double, 3, 2, 0> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) const
==11802==ABORTING
```

christophebedard, 2025-03-26:
Thanks for reporting this. Did you figure out how/why it happens? Would you be willing to submit a PR to fix it? 😁

zhihaoshang, 2025-03-27:
> 感谢您报告此事。你弄清楚它是如何/为什么发生的吗？你愿意提交一个 PR 来修复它吗？😁

Thank you for your feedback! Based on the logs and stack information, the root cause of the issue is that the transformation object passed in is either null or not properly initialized, causing the transform function to attempt reading a null pointer. This is a logical flaw within the function under test. To avoid this issue, appropriate null pointer checks should be added before and after the function call to ensure the validity of the relevant objects.

christophebedard, 2025-03-27:
It would be helpful if you could provide a minimal example that only contains the code that reproduces the issue. I'm having a hard time finding the relevant part of the sample you provided.

zhihaoshang, 2025-03-29:
> 如果您可以提供一个仅包含重现问题的代码的最小示例，将会很有帮助。我很难在您提供的样本中找到相关部件。

Thank you for your response. I understand your request. Below is a minimal unit test case that reproduces the issue where tf2_ros::BufferInterface::transform triggers an AddressSanitizer (SEGV) error.
```cpp
// To get M_PI, especially on Windows.
#ifdef _MSC_VER
#ifndef _USE_MATH_DEFINES
#define _USE_MATH_DEFINES
#endif
#endif
#include <cmath>
#include <memory>
// Version 3.4.0 of Eigen in Ubuntu 22.04 has a bug that causes -Wclass-memaccess warnings on
// aarch64.  Upstream Eigen has already fixed this in
// https://gitlab.com/libeigen/eigen/-/merge_requests/645 .  The Debian fix for this is in
// https://salsa.debian.org/science-team/eigen3/-/merge_requests/1 .
// However, it is not clear that that fix is going to make it into Ubuntu 22.04 before it
// freezes, so disable the warning here.
#if defined(__GNUC__) && !defined(__clang__)
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wclass-memaccess"
#endif
#include <Eigen/Geometry>  // NOLINT
#if defined(__GNUC__) && !defined(__clang__)
#pragma GCC diagnostic pop
#endif
#include "gtest/gtest.h"
#include "geometry_msgs/msg/point.hpp"
#include "geometry_msgs/msg/point_stamped.hpp"
#include "geometry_msgs/msg/pose.hpp"
#include "geometry_msgs/msg/pose_stamped.hpp"
#include "geometry_msgs/msg/transform_stamped.hpp"
#include "rclcpp/clock.hpp"
#include "tf2/convert.hpp"
#include "tf2/transform_datatypes.hpp"
#include "tf2_eigen/tf2_eigen.hpp"
#include "tf2_ros/buffer.h"
#include "tf2_ros/transform_listener.h"

struct EigenBufferTransform_v2 : public ::testing::Test
{
  static void SetUpTestSuite_v2()
  {
    rclcpp::Clock::SharedPtr clock = std::make_shared<rclcpp::Clock>(RCL_SYSTEM_TIME);
    tf_buffer = std::make_unique<tf2_ros::Buffer>(clock);
    tf_buffer->setUsingDedicatedThread(true);
    // populate buffer
    geometry_msgs::msg::TransformStamped t;
    t.transform.translation.x = 10;
    t.transform.translation.y = 20;
    t.transform.translation.z = 30;
    t.transform.rotation.w = 0;
    t.transform.rotation.x = 1;
    t.transform.rotation.y = 0;
    t.transform.rotation.z = 0;
    t.header.stamp = tf2_ros::toMsg(tf2::timeFromSec(2));
    t.header.frame_id = "A";
    t.child_frame_id = "B";
    tf_buffer->setTransform(t, "test");
  }
  template<int mode>
  void testEigenTransform_v2();
  ::testing::AssertionResult doTestEigenQuaternion_v2(
    const Eigen::Quaterniond & parameter, const Eigen::Quaterniond & expected);
  static std::unique_ptr<tf2_ros::Buffer> tf_buffer;
  static constexpr double EPS = 1e-3;
};

std::unique_ptr<tf2_ros::Buffer> EigenBufferTransform_v2::tf_buffer;

template<int mode>
void EigenBufferTransform_v2::testEigenTransform_v2()
{
  using T = Eigen::Transform<double, 3, mode>;
  using stampedT = tf2::Stamped<T>;
  const stampedT i1{
    T{Eigen::Translation3d{1, 2, 3} *Eigen::Quaterniond{0, 1, 0, 0}}, tf2::timeFromSec(2), "A"};
  // simple api
  const stampedT i_simple = tf_buffer->transform(i1, "B", tf2::durationFromSec(2.0));
  EXPECT_NEAR(i_simple.translation().x(), -9, EPS);
  EXPECT_NEAR(i_simple.translation().y(), 18, EPS);
  EXPECT_NEAR(i_simple.translation().z(), 27, EPS);
  const auto q1 = Eigen::Quaterniond(i_simple.linear());
  EXPECT_NEAR(q1.x(), 0.0, EPS);
  EXPECT_NEAR(q1.y(), 0.0, EPS);
  EXPECT_NEAR(q1.z(), 0.0, EPS);
  EXPECT_NEAR(q1.w(), 1.0, EPS);
  // advanced api
  const stampedT v_advanced =
    tf_buffer->transform(i1, "B", tf2::timeFromSec(2.0), "A", tf2::durationFromSec(3.0));
  EXPECT_NEAR(v_advanced.translation().x(), -9, EPS);
  EXPECT_NEAR(v_advanced.translation().y(), 18, EPS);
  EXPECT_NEAR(v_advanced.translation().z(), 27, EPS);
  const auto q2 = Eigen::Quaterniond(v_advanced.linear());
  EXPECT_NEAR(q2.x(), 0.0, EPS);
  EXPECT_NEAR(q2.y(), 0.0, EPS);
  EXPECT_NEAR(q2.z(), 0.0, EPS);
  EXPECT_NEAR(q2.w(), 1.0, EPS);
}

TEST_F(EigenBufferTransform_v2, Affine3d_v2) {
  testEigenTransform_v2<Eigen::Affine>();
}
```
Please take a look at this minimal reproducible code and let me know if you need any further details. Thank you!

MarcoMatteoBassa, 2025-04-03:
I just got a similar issue after trying to transform a pose from a costmap layer... however it might be unrelated.
Could you try to add an

`#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"`

Also, how does your CMakeLists look like?

zhihaoshang, 2025-04-03:
> 在尝试从 costmap 图层转换姿势后，我刚刚遇到了类似的问题......但是，它可能无关。您能否尝试添加一个
> 
> `#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"`
> 
> 另外，你的 CMakeLists 是什么样的？

Thank you for your reply!
My CMakeLists looks like this:
```cpp
cmake_minimum_required(VERSION 3.5)
project(tf2_eigen)

# Default to C++17
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 17)
  set(CMAKE_CXX_STANDARD_REQUIRED ON)
endif()
if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic -Wnon-virtual-dtor -Woverloaded-virtual)
endif()

find_package(ament_cmake REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(tf2 REQUIRED)
find_package(tf2_ros REQUIRED)

# Work around broken find module in AlmaLinux/RHEL eigen3-devel from PowerTools repo
find_package(Eigen3 QUIET NO_MODULE)
if(NOT Eigen3_FOUND)
  find_package(Eigen3 REQUIRED)
endif()

add_library(tf2_eigen INTERFACE)
target_link_libraries(tf2_eigen INTERFACE
  ${geometry_msgs_TARGETS}
  tf2::tf2
  tf2_ros::tf2_ros)
target_include_directories(tf2_eigen INTERFACE
  "$<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>"
  "$<INSTALL_INTERFACE:include/${PROJECT_NAME}>")
if(TARGET Eigen3::Eigen)
  # TODO(sloretz) require target to exist when https://github.com/ros2/choco-packages/issues/19 is addressed
  target_link_libraries(tf2_eigen INTERFACE Eigen3::Eigen)
else()
  target_include_directories(tf2_eigen INTERFACE ${Eigen3_INCLUDE_DIRS})
endif()

install(TARGETS tf2_eigen EXPORT export_tf2_eigen)
install(DIRECTORY include/ DESTINATION include/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_cmake_cppcheck REQUIRED)
  find_package(ament_cmake_cpplint REQUIRED)
  find_package(ament_cmake_lint_cmake REQUIRED)
  find_package(ament_cmake_uncrustify REQUIRED)

  ament_cppcheck(LANGUAGE "c++")
  ament_cpplint()
  ament_lint_cmake()
  ament_uncrustify(LANGUAGE "c++")

  find_package(ament_cmake_gtest REQUIRED)

  ament_add_gtest(tf2_eigen-test test/tf2_eigen-test.cpp)
  if(TARGET tf2_eigen-test)
    target_link_libraries(tf2_eigen-test
      tf2_eigen
      # Used, but not linked to test tf2_eigen's exports:
      #   Eigen3::Eigen
      #   ${geometry_msgs_TARGETS}
      #   tf2::tf2
      #   tf2_ros::tf2_ros
    )
  endif()
endif()

ament_export_dependencies(
  Eigen3
  tf2
  tf2_ros)

# Export old-style CMake variables
ament_export_include_directories("include/${PROJECT_NAME}")

# Export modern CMake targets
ament_export_targets(export_tf2_eigen)

ament_package()
```

MarcoMatteoBassa, 2025-04-07:
I think your test, as it is written, is not initializing the fixture class. Explicitally calling SetUpTestSuite_v2 makes it work.
You can check again how to setup test-Fixture classes here https://github.com/google/googletest/blob/main/docs/primer.md

christophebedard, 2025-04-16:
Good catch, thank you! Changing `SetUpTestSuite_v2()` to `SetUpTestSuite()` makes the test pass for me as well.

Since it looks like this is not an actual bug, I'll close this. Feel free to keep discussing if you think this is a bug.

closed 2025-04-16 by christophebedard