ros2/geometry2 #766: BufferCore::cancelTransformableRequest cancels unrelated callbacks

roehling, 2025-03-14:
As mentioned in [this comment](https://github.com/ros2/geometry2/pull/475#issuecomment-2724836575), the cancellation of transformable requests can have unintended destructive side effects and also cancel unrelated callbacks. The following C++ program reproduces the issue:

```c++
#include <rclcpp/time.hpp>
#include <tf2/buffer_core.hpp>

#include <iostream>

void callback(tf2::TransformableRequestHandle request_handle, const std::string& target_frame,
              const std::string& source_frame, const tf2::TimePoint& time, tf2::TransformableResult result)
{
    std::cout << "callback(request_handle=" << request_handle << ", target_frame=\"" << target_frame
              << "\", source_frame=\"" << source_frame << "\", time=" << tf2::displayTimePoint(time)
              << ", result=" << result << ")\n";
}

template<typename Rep, typename Period>
tf2::TransformableRequestHandle add_transformable_request(tf2::BufferCore& bc, const std::string& target_frame,
                                                          const std::string& source_frame,
                                                          const std::chrono::duration<Rep, Period>& time)
{
    tf2::TimePoint tp{time};
    tf2::TransformableRequestHandle h = bc.addTransformableRequest(callback, target_frame, source_frame, tp);
    std::cout << "add_transformable_request(target_frame=\"" << target_frame << "\", source_frame=\"" << source_frame
              << "\", time=" << tf2::displayTimePoint(tp) << ") -> " << h << "\n";
    return h;
}

template<typename Rep, typename Period>
void set_transform(tf2::BufferCore& bc, const std::string& target_frame, const std::string& source_frame,
                   const std::chrono::duration<Rep, Period>& time)
{
    std::chrono::nanoseconds ns = std::chrono::duration_cast<std::chrono::nanoseconds>(time);
    geometry_msgs::msg::TransformStamped tr;
    tr.header.frame_id = target_frame;
    tr.header.stamp = rclcpp::Time(ns.count());
    tr.child_frame_id = source_frame;
    tr.transform.rotation.w = 1;  // make transform valid
    std::cout << "set_transform(target_farme=\"" << target_frame << "\", source_frame=\"" << source_frame
              << "\", time=" << tf2::displayTimePoint(tf2::TimePoint(time)) << ")\n";
    bc.setTransform(tr, "test_injector", false);
}

void cancel_transformable_request(tf2::BufferCore& bc, tf2::TransformableRequestHandle request_handle)
{
    std::cout << "cancel_transformable_request(request_handle=" << request_handle << ")\n";
    bc.cancelTransformableRequest(request_handle);
}

int main()
{
    using namespace std::chrono_literals;
    tf2::BufferCore bc{10s};
    tf2::TransformableRequestHandle handle = add_transformable_request(bc, "target", "alpha", 100s);
    add_transformable_request(bc, "target", "beta", 100s);
    add_transformable_request(bc, "target", "gamma", 100s);
    cancel_transformable_request(bc, handle);
    set_transform(bc, "target", "alpha", 100s);
    set_transform(bc, "target", "beta", 100s);
    set_transform(bc, "target", "gamma", 100s);
    return 0;
}
```

The program output is as follows:
```
add_transformable_request(target_frame="target", source_frame="alpha", time=100.000000) -> 1
add_transformable_request(target_frame="target", source_frame="beta", time=100.000000) -> 2
add_transformable_request(target_frame="target", source_frame="gamma", time=100.000000) -> 3
cancel_transformable_request(request_handle=1)
set_transform(target_farme="target", source_frame="alpha", time=100.000000)
set_transform(target_farme="target", source_frame="beta", time=100.000000)
callback(request_handle=2, target_frame="target", source_frame="beta", time=100.000000, result=0)
set_transform(target_farme="target", source_frame="gamma", time=100.000000)
```

As you can see, the cancellation of the `alpha` request inadvertently also destroyed the callback for the `gamma` request.

roehling, 2025-03-14:
For reference, [this comment](https://github.com/ros2/geometry2/pull/475#issuecomment-965670344) pinpoints the root cause: `std::remove_if()` does not preserve any deleted entries, so `cb_handle` can no longer be trusted to contain the correct value.

fujitatomoya, 2025-03-27:
this does look like an issue, please consider pull request.

PR #779 (2025-04-10): Do not clobber callback handles when cancelling pending transformable requests
Fixes #766 

This PR also applies cleanly to the `humble` and `jazzy` branches; I'd appreciate a backport.
closed 2025-04-10 by ahcorde