ros2/geometry2 #768: Data Overwrite Issue in StaticCache Class's insertData and getData Methods

zhihaoshang, 2025-03-21:
# Bug Report
## Environment
OS Version: Ubuntu 24.04
geometry2 version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
In the current implementation, the insertData method of the StaticCache class overwrites the internally stored TransformStorage object every time new data is inserted. As a result, the getData method always returns the most recently inserted transform data, ignoring the time stamp provided in the query.

When using the StaticCache class, if multiple pieces of transform data are inserted with different time stamps, the getData method always returns the most recent data, which leads to functional errors, particularly in scenarios where data needs to be retrieved based on a specific time stamp.

## Steps to reproduce
```cpp
#include <gtest/gtest.h>
#include <chrono>
#include <cmath>
#include <stdexcept>
#include <tf2/time_cache.hpp>

void resetStorage(tf2::TransformStorage & stor)
{
  stor.translation_.setValue(1.0, -1.0, 0.5);
  stor.rotation_.setValue(0.707, 0.0, 0.707, 0.0);
}

TEST(StaticCache, LargeVolumeInsertionEdgeCases)
{
  tf2::StaticCache cache;
  tf2::TransformStorage stor;
  resetStorage(stor);
  
  for (int i = 0; i < 5000; ++i) {
    stor.frame_id_ = tf2::CompactFrameID(i);
    stor.stamp_ = tf2::TimePoint(std::chrono::nanoseconds(i));
    EXPECT_NO_THROW(cache.insertData(stor));
  }

  tf2::TransformStorage result;
  EXPECT_TRUE(cache.getData(tf2::TimePoint(std::chrono::nanoseconds(2500)), result));
  EXPECT_EQ(result.frame_id_, 2500);
}
int main(int argc, char ** argv)
{
  testing::InitGoogleTest(&argc, argv);
  return RUN_ALL_TESTS();
}
```

## Output
Here is what the output shows:
```markdown
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from StaticCache
[ RUN      ] StaticCache.LargeVolumeInsertionEdgeCases
/home/shangzh/geometry2_ws/geometry2/tf2/test/static_cache_test.cpp:27: Failure
Expected equality of these values:
  result.frame_id_
    Which is: 4999
  2500

[  FAILED  ] StaticCache.LargeVolumeInsertionEdgeCases (0 ms)
[----------] 1 test from StaticCache (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] StaticCache.LargeVolumeInsertionEdgeCases

 1 FAILED TEST
```

fujitatomoya, 2025-04-03:
please consider pull request for this! thanks!

InvincibleRMC, 2025-04-11:
Looking into this it seems to be an intentional design decision to have `StaticCache` only store one `TransformStorage`. Using `TimeCache` instead seems to be what you want.

```c++
#include <gtest/gtest.h>
#include <chrono>
#include <cmath>
#include <stdexcept>
#include <tf2/time_cache.hpp>

void resetStorage(tf2::TransformStorage & stor)
{
  stor.translation_.setValue(1.0, -1.0, 0.5);
  stor.rotation_.setValue(0.707, 0.0, 0.707, 0.0);
}

TEST(TimeCache, LargeVolumeInsertionEdgeCases)
{
  tf2::TimeCache cache;
  tf2::TransformStorage stor;
  resetStorage(stor);
  
  for (int i = 0; i < 5000; ++i) {
    stor.frame_id_ = tf2::CompactFrameID(i);
    stor.stamp_ = tf2::TimePoint(std::chrono::nanoseconds(i));
    EXPECT_NO_THROW(cache.insertData(stor));
  }

  tf2::TransformStorage result;
  EXPECT_TRUE(cache.getData(tf2::TimePoint(std::chrono::nanoseconds(2500)), result));
  EXPECT_EQ(result.frame_id_, 2500);
}
int main(int argc, char ** argv)
{
  testing::InitGoogleTest(&argc, argv);
  return RUN_ALL_TESTS();
}
```

kscottz, 2025-04-11:
@InvincibleRMC you want to give it a shot?

InvincibleRMC, 2025-04-11:
> @InvincibleRMC you want to give it a shot?

I looked into it but, I'm not sure it is a bug in the StaticCache. It seems that TimeChache is the class that should be used instead. Trying their example test case with it makes it pass. I was unable to find documentation clarifying the distinction between them though.

kscottz, 2025-04-14:
Ah, so what you're saying is that:

1) The intended behaviors for `StaticCache` and `TimeCache` differ and are not equivalent. 
2) The bug as described is the intended behavior for `StaticCache.`

Sounds like we should close the issue as "won't fix." @fujitatomoya that sound good to you?

kscottz, 2025-04-14:
Also @InvincibleRMC thanks for the help!

zhihaoshang, 2025-04-15:
> 请考虑为此提交 Pull Request！谢谢！

Thank you for your response!

I would appreciate it if you could submit the PR for the fix, as I’m currently not familiar with the process of submitting a PR myself. Your expertise in this area would ensure that the fix is implemented correctly and efficiently.

Please let me know if you need any additional information from my side to move forward with the PR.

Thank you again for your help!

InvincibleRMC, 2025-04-15:
Can you try using `TimeCache` instead of `StaticCache`?

fujitatomoya, 2025-04-15:
@zhihaoshang 

what do you think above suggestion?

besides that, according to the history, are you using GenAI or any static tools to create the issue? can you share what kind of tools are you using to create issue and response?

Reference: https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

closed 2025-04-16 by zhihaoshang