ros2/geometry2 #769: tf2::StaticCache::getData() Returns true Even When Cache is Empty or Timestamp Mismatches

zhihaoshang, 2025-03-21:
# Bug Report
## Environment
OS Version: Ubuntu 24.04
geometry2 version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
In tf2::StaticCache, the behavior of the getData() method does not match expectations. When the cache is empty or the queried timestamp does not match the stored timestamp, getData() still returns true instead of false. This may cause the caller to retrieve incorrect transformation data, potentially affecting coordinate transformations.

## Steps to reproduce
```cpp
#include <gtest/gtest.h>
#include <chrono>
#include <cmath>
#include <stdexcept>
#include <tf2/time_cache.hpp>

void resetStorage(tf2::TransformStorage & stor)
{
  stor.translation_.setValue(1.0, -1.0, 0.5);
  stor.rotation_.setValue(0.707, 0.0, 0.707, 0.0);
}

TEST(StaticCache, EmptyCacheRetrieval)
{
  tf2::StaticCache cache;
  tf2::TransformStorage stor;
  resetStorage(stor);
  EXPECT_FALSE(cache.getData(tf2::TimePoint(std::chrono::nanoseconds(99999)), stor));
}

TEST(StaticCache, ErrorHandlingForInvalidTimestamps)
{
  tf2::StaticCache cache;
  tf2::TransformStorage stor;
  resetStorage(stor);
  stor.frame_id_ = tf2::CompactFrameID(10);
  stor.stamp_ = tf2::TimePoint(std::chrono::nanoseconds(10));
  
  cache.insertData(stor);
  tf2::TransformStorage result;
  EXPECT_FALSE(cache.getData(tf2::TimePoint(std::chrono::nanoseconds(9999)), result));
}

int main(int argc, char ** argv)
{
  testing::InitGoogleTest(&argc, argv);
  return RUN_ALL_TESTS();
}

```
## Output
```markdown
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from StaticCache
[ RUN      ] StaticCache.EmptyCacheRetrieval
/home/shangzh/geometry2_ws/geometry2/tf2/test/static_cache_test.cpp:18: Failure
Value of: cache.getData(tf2::TimePoint(std::chrono::nanoseconds(99999)), stor)
  Actual: true
Expected: false

[  FAILED  ] StaticCache.EmptyCacheRetrieval (0 ms)
[ RUN      ] StaticCache.ErrorHandlingForInvalidTimestamps
/home/shangzh/geometry2_ws/geometry2/tf2/test/static_cache_test.cpp:31: Failure
Value of: cache.getData(tf2::TimePoint(std::chrono::nanoseconds(9999)), result)
  Actual: true
Expected: false

[  FAILED  ] StaticCache.ErrorHandlingForInvalidTimestamps (0 ms)
[----------] 2 tests from StaticCache (0 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] StaticCache.EmptyCacheRetrieval
[  FAILED  ] StaticCache.ErrorHandlingForInvalidTimestamps

 2 FAILED TESTS

```

fujitatomoya, 2025-04-03:
please consider pull request for this! thanks!

zhihaoshang, 2025-04-15:
> 请考虑为此提交 Pull Request！谢谢！

Thank you for your response!

I would appreciate it if you could submit the PR for the fix, as I’m currently not familiar with the process of submitting a PR myself. Your expertise in this area would ensure that the fix is implemented correctly and efficiently.

Please let me know if you need any additional information from my side to move forward with the PR.

Thank you again for your help!

maxkeller321, 2025-04-26:
Is this issue still relevant @fujitatomoya ?

kscottz, 2025-05-01:
@maxkeller321 yes, please take a look. This issue is from a static analyzer so it may not need to be addressed.