ros2/geometry2 #770: Overflow Issue in durationFromSec() Function when Handling Extremely Large or Small Values

zhihaoshang, 2025-03-23:
# Bug Report
## Environment
OS Version: Ubuntu 24.04
geometry2 version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
The tf2::durationFromSec() function encounters an overflow or precision loss issue when processing extremely large (e.g., 1e308) or extremely small (e.g., -1e308) floating-point second values. The implementation of this function uses int32_t to represent seconds and nanoseconds, but the range of int32_t is insufficient to handle such large values. Due to this limitation, when the input t_sec exceeds the range of int32_t, the result overflows, leading to invalid time values being returned.

## Steps to reproduce
```cpp
#include <gtest/gtest.h>
#include <chrono>
#include "tf2/time.hpp"
using namespace std::literals::chrono_literals;

TEST(TestTime, durationFromSec_LargeNegativeValues) {
  EXPECT_EQ(tf2::durationFromSec(-1e308), -1e308 * 1s); // Extremely large negative values
}

TEST(TestTime, durationFromSec_LargePositiveValues) {
  EXPECT_EQ(tf2::durationFromSec(1e308), 1e308 * 1s); // Extremely large positive values
}
```
## Output
```marksodn
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from TestTime
[ RUN      ] TestTime.durationFromSec_LargeNegativeValues
/home/shangzh/geometry2_ws/geometry2/tf2/test/test_time.cpp:7: Failure
Expected equality of these values:
  tf2::durationFromSec(-1e308)
    Which is: 8-byte object <00-00 00-80 FF-64 CD-1D>
  -1e308 * 1s
    Which is: 8-byte object <A0-C8 EB-85 F3-CC E1-FF>

[  FAILED  ] TestTime.durationFromSec_LargeNegativeValues (0 ms)
[ RUN      ] TestTime.durationFromSec_LargePositiveValues
/home/shangzh/geometry2_ws/geometry2/tf2/test/test_time.cpp:11: Failure
Expected equality of these values:
  tf2::durationFromSec(1e308)
    Which is: 8-byte object <00-00 00-80 FF-64 CD-1D>
  1e308 * 1s
    Which is: 8-byte object <A0-C8 EB-85 F3-CC E1-7F>

[  FAILED  ] TestTime.durationFromSec_LargePositiveValues (0 ms)
[----------] 2 tests from TestTime (0 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] TestTime.durationFromSec_LargeNegativeValues
[  FAILED  ] TestTime.durationFromSec_LargePositiveValues

 2 FAILED TESTS

```

cramke, 2025-04-13:
What is the expected behavior for such an overflow? 

The input is double (10^308), the function body is int32_t (10^9) and the return value tf2::Duration is int64_t (10^18). Throw a runtime exception, clamp the double and continue, increase the size of tf2::Duration ....?

kscottz, 2025-04-14:
That's a great question @cramke. In my mind clamping / extending a duration had quite similar effects where the duration of an event would either last practically indefinitely or return almost instantly. I guess there's a fourth option of clamping the values and then issuing a warning message. 

Personally, I think throwing a runtime exception is probably the best option here. If you are asking for an indefinitely long or infinitesimally short duration you are probably doing something wrong upstream of this function. @fujitatomoya do you have any thoughts here?

fujitatomoya, 2025-04-16:
this is a bug, at least not well-handled those numbers.
but i think we already know this behavior??? (i am not familiar with tf2 history, but there are some test cases for those conditions.)

https://github.com/ros2/geometry2/blob/7767a7072e7fb9acb7d18127984fccee048de060/tf2/test/test_time.cpp#L48-L49

i think that throwing `std::overflow_error` sounds the best option here. or maybe we can consider this in `rcpputils` instead.

cramke, 2025-04-22:
I opened a Pull Request to throw an overflow_error for values too large or too small.

fujitatomoya, 2025-04-23:
> but i think we already know this behavior??? (i am not familiar with tf2 history, but there are some test cases for those conditions.)

sorry this is not correct, they are going to be changed into zero but those stay in the range of int32_t.

cramke, 2025-04-23:
I think we are mixing two similar issues here?

The test you mentioned is for a number which is close to zero but not exactly zero (1e-40). For me locally this test is passing, so I can change the behavior but I don't see an issue with them at the moment. 

The original test of this issue was about a number which is absolut very large (-1e200). These new failing tests are currently not part of the rolling branch.

PR #787 (2025-04-24): Overflow Issue in durationFromSec() Function when Handling Extremely Large or Small Values (backport #785)
Closes #770 

Adds a check to throw an overflow_error to avoid failing silently when input is extremly large or small. <hr>This is an automatic backport of pull request #785 done by [Mergify](https://mergify.com).
PR #786 (2025-04-24): Overflow Issue in durationFromSec() Function when Handling Extremely Large or Small Values (backport #785)
Closes #770 

Adds a check to throw an overflow_error to avoid failing silently when input is extremly large or small. <hr>This is an automatic backport of pull request #785 done by [Mergify](https://mergify.com).