ros2/geometry2 #771: operator== Comparison Logic Issue in TransformStorage

zhihaoshang, 2025-03-23:
# Bug Report
## Environment
OS Version: Ubuntu 24.04
geometry2 version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
In the TransformStorage class, the operator== is used to compare the member variables of two TransformStorage objects. However, the current implementation does not account for cases where some members (e.g., frame_id_) may be in an uninitialized or default-constructed state. This can lead to incorrect comparison results. Specifically, when a member variable has a default-constructed value, it is mistakenly considered equal, causing the comparison to yield unexpected results.

## Steps to reproduce
```cpp
#include <gtest/gtest.h>
#include "tf2/LinearMath/Quaternion.hpp"
#include "tf2/LinearMath/Vector3.hpp"
#include "tf2/time.hpp"
#include "tf2/transform_storage.hpp"

class TransformStorageTest : public ::testing::Test
{
protected:
  tf2::TransformStorage createTransformStorage()
  {
    const tf2::CompactFrameID frame_id(0);
    const tf2::CompactFrameID child_frame_id(1);
    const tf2::TimePoint stamp(tf2::TimePointZero);
    const tf2::Quaternion rotation(0.0, 0.0, 0.0, 1.0);
    const tf2::Vector3 translation(0.0, 0.0, 0.0);
    return tf2::TransformStorage(stamp, rotation, translation, frame_id, child_frame_id);
  }
};

TEST_F(TransformStorageTest, InvalidComparison_UninitializedMemberImpact) {
  tf2::TransformStorage t1 = createTransformStorage();
  tf2::TransformStorage t2 = t1;
  t2.frame_id_ = tf2::CompactFrameID();  // potential uninitialized/invalid value
  ASSERT_FALSE(t1 == t2);
}

```
## Output
```markdown
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TransformStorageTest
[ RUN      ] TransformStorageTest.InvalidComparison_UninitializedMemberImpact
/home/shangzh/geometry2_ws/geometry2/tf2/test/test_storage.cpp:25: Failure
Value of: t1 == t2
  Actual: true
Expected: false

[  FAILED  ] TransformStorageTest.InvalidComparison_UninitializedMemberImpact (0 ms)
[----------] 1 test from TransformStorageTest (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TransformStorageTest.InvalidComparison_UninitializedMemberImpact

 1 FAILED TEST
```

fujitatomoya, 2025-04-03:
this looks like a bug/problem, can you consider the pull request?

andschi, 2025-04-13:
I had a look at this. `tf2::CompactFrameID` is an alias for `uint32_t`. In `createTransformStorage()`, `frame_id` is initialized with `0`. 

This means, the reported issue comes down to
```
uint32_t frame_id_1 = 0; 
uint32_t frame_id_2 = 0; 
frame_id_1 = uint32_t(); // Value of default-constructed uint32_t may still be 0. 
ASSERT_FALSE(frame_id_1 == frame_id_2); // fails as 0 == 0
```

My question is if 0 is a valid `frame_id` / `child_frame_id`? If not, I would propose to validate that the frame ids are non-zero. Otherwise, the comparison should return false. I opened a PR with this fix.

kscottz, 2025-04-14:
Hi @andschi 

Thanks for the help. 

Good question. Doing a cursory check of the source [it appears the tests use tf2:CompactFrameID with a value of zero](https://github.com/search?q=repo%3Aros2%2Fgeometry2%20CompactFrameID&type=code) and they aren't testing for that particular case. This makes me believe that zero is a valid frame id and that the comparison should return false. I don't know a ton about geometry2 internals so take this with a grain of salt.

kscottz, 2025-04-17:
> If zero is indeed a valid CompactFrameID, I would propose to default initialize the frame_id_ and the child_frame_id_ in the TransformStorage to UINT32_MAX (which I hope is deemed invalid). At least then, they do not hold something valid until they are explicitly set.

That sounds fairly trivial. Let's go ahead and do that and close this out. We believe that this issue came in from some sort of automated tool so it is probably overly pedantic, but this change would clear out the warning for the user.

andschi, 2025-04-22:
Hi @kscottz , I hope you had a great extended Easter weekend! 

I revised the [pull request](https://github.com/ros2/geometry2/pull/783) to the agreed `default initialize the frame_id_ and the child_frame_id_ in the TransformStorage to UINT32_MAX`. 

I have not been part of the merging process before. How is this put forward? I am under Ubuntu22 and created the PR for the `humble` branch. Who will cherry-pick the commit for the other ros2 releases?

fujitatomoya, 2025-04-25:
@andschi @kscottz 

thanks for working on this.

coming back to look at the original issue report, it does not really make sense. and that is explained in https://github.com/ros2/geometry2/issues/771#issuecomment-2799844824
having the default initialization value seems to be fine though until user application sets that.

@zhihaoshang can you explain why this is the problem for ros2/geometry2?

fujitatomoya, 2025-04-29:
@zhihaoshang friendly ping

fujitatomoya, 2025-05-19:
@zhihaoshang can you share your though about https://github.com/ros2/geometry2/issues/771#issuecomment-2831343452?

christophebedard, 2025-05-31:
Looks like we may not get an answer, so I think we should just go with what we think is best (and otherwise maybe err on the side of caution/conservativeness?).

fujitatomoya, 2025-06-14:
@andschi let's have it merged and close this. thanks for the ping.

closed 2025-06-15 by fujitatomoya