ros2/geometry2 #773: Import error with sensor_msgs

ryan-roche, 2025-03-27:
### Operating System:

Ubuntu 22.04

### ROS version or commit hash:

Humble

### RMW Configuration (if applicable):

The python `sensor_msgs` module containing the `read_points` and `create_cloud` functions that `tf2_sensor_msgs.py` imports has been renamed to `sensor_msgs_py`, so the package currently fails to import those functions 

### 'ros2 doctor --report' output

```Formatted

```

### Steps to reproduce issue

1. Install the `ros-humble-python-tf2-sensor-msgs` package
2. Attempt to import `tf2_sensor_msgs` in a python file

### Expected behavior

The module should import noramlly

### Actual behavior

The module throws an error as it cannot find the `sensor_msgs.point_cloud2` python module, as it is now `sensor_msgs_py.point_cloud2`

ryan-roche, 2025-03-27:
I would, but the problem is deeper than that. Changing the import name from `sensor_msgs` to `sensor_msgs_py` fixes _that_ error, but once it gets through importing those functions it runs into another import error:

```
[locate_object-13] Traceback (most recent call last):
[locate_object-13]   File "/ros_ws/install/spot_autogrammetry/lib/spot_autogrammetry/locate_object", line 33, in <module>
[locate_object-13]     sys.exit(load_entry_point('spot-autogrammetry', 'console_scripts', 'locate_object')())
[locate_object-13]   File "/ros_ws/install/spot_autogrammetry/lib/spot_autogrammetry/locate_object", line 25, in importlib_load_entry_point
[locate_object-13]     return next(matches).load()
[locate_object-13]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[locate_object-13]     module = import_module(match.group('module'))
[locate_object-13]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[locate_object-13]     return _bootstrap._gcd_import(name[level:], package, level)
[locate_object-13]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[locate_object-13]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[locate_object-13]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[locate_object-13]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[locate_object-13]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[locate_object-13]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[locate_object-13]   File "/ros_ws/build/spot_autogrammetry/spot_autogrammetry/locate_object.py", line 14, in <module>
[locate_object-13]     from tf2_sensor_msgs.tf2_sensor_msgs import do_transform_cloud
[locate_object-13]   File "/usr/lib/python3/dist-packages/tf2_sensor_msgs/__init__.py", line 1, in <module>
[locate_object-13]     from .tf2_sensor_msgs import *
[locate_object-13]   File "/usr/lib/python3/dist-packages/tf2_sensor_msgs/tf2_sensor_msgs.py", line 31, in <module>
[locate_object-13]     import rospy
[locate_object-13]   File "/usr/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>
[locate_object-13]     from .client import spin, myargv, init_node, \
[locate_object-13]   File "/usr/lib/python3/dist-packages/rospy/client.py", line 54, in <module>
[locate_object-13]     import rospy.core
[locate_object-13]   File "/usr/lib/python3/dist-packages/rospy/core.py", line 75, in <module>
[locate_object-13]     from rosgraph_msgs.msg import Log
[locate_object-13] ImportError: cannot import name 'Log' from 'rosgraph_msgs.msg' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosgraph_msgs/msg/__init__.py)
```

`rospy` isn't a thing in ROS2- it's been replaced by `rclpy`. AFAIK the Log message isn't even a part of the `rclpy` package, instead being inside `rcl_interfaces`. Unless the packages are a direct drop-in, which I'll check, this may be a more involved fix

ryan-roche, 2025-03-27:
Ok, you can _literally_ just remove the `import rospy` line and it works. I'm getting an error when I try to convert a pointcloud message-
```
[locate_object-13] Traceback (most recent call last):
[locate_object-13]   File "/ros_ws/install/spot_autogrammetry/lib/spot_autogrammetry/locate_object", line 33, in <module>
[locate_object-13]     sys.exit(load_entry_point('spot-autogrammetry', 'console_scripts', 'locate_object')())
[locate_object-13]   File "/ros_ws/build/spot_autogrammetry/spot_autogrammetry/locate_object.py", line 123, in main
[locate_object-13]     rclpy.spin(service_node)
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 226, in spin
[locate_object-13]     executor.spin_once()
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 739, in spin_once
[locate_object-13]     self._spin_once_impl(timeout_sec)
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 736, in _spin_once_impl
[locate_object-13]     raise handler.exception()
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[locate_object-13]     self._handler.send(None)
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 437, in handler
[locate_object-13]     await call_coroutine(entity, arg)
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 391, in _execute_service
[locate_object-13]     response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
[locate_object-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[locate_object-13]     return callback(*args)
[locate_object-13]   File "/ros_ws/build/spot_autogrammetry/spot_autogrammetry/locate_object.py", line 84, in handle_service
[locate_object-13]     gpe_points = self.tf_buffer.transform(points, 'gpe')
[locate_object-13]   File "/opt/ros/humble/lib/python3.10/site-packages/tf2_ros/buffer_interface.py", line 96, in transform
[locate_object-13]     res = do_transform(object_stamped, self.lookup_transform(target_frame, object_stamped.header.frame_id,
[locate_object-13]   File "/usr/lib/python3/dist-packages/tf2_sensor_msgs/tf2_sensor_msgs.py", line 57, in do_transform_cloud
[locate_object-13]     points_out.append((p_out[0], p_out[1], p_out[2]) + p_in[3:])
[locate_object-13] IndexError: too many indices for array: array is 0-dimensional, but 1 were indexed
```
but that might be unrelated.

I'll submit a PR for the import errors, at least

christophebedard, 2025-03-27:
Good point: https://github.com/ros2/geometry2/blob/1417e78528a0f69d05c3a8aaa245d296e807f304/tf2_sensor_msgs/src/tf2_sensor_msgs/tf2_sensor_msgs.py#L32

So some of the changes made between `tf2_sensor_msgs/src/tf2_sensor_msgs/tf2_sensor_msgs.py` and `tf2_sensor_msgs/tf2_sensor_msgs/tf2_sensor_msgs.py` in https://github.com/ros2/geometry2/pull/507 need to be ported to Humble. No

ryan-roche, 2025-03-27:
I was able to just remove that line altogether and I was able to import tf2_sensor_msgs without error. Still unsure what was causing that error with the PointCloud2 conversion, but the PR I submitted fixed the import errors

ryan-roche, 2025-03-27:
Shoot- lemme rebase the commits and sign off on them

christophebedard, 2025-03-28:
As discussed in #774 (e.g., https://github.com/ros2/geometry2/pull/774#issuecomment-2759847519), the issue with `tf2_sensor_msgs.py` is a bit larger than a simple import error: #507 should be backported to Humble. I've created #775 for that. Let me know if you'd like to contribute a PR for that.

For now, I'll close this issue. Thanks again for reporting this!

closed 2025-03-28 by christophebedard