ros2/geometry2 #788: # Issue: tf2_ros::MessageFilter blocks indefinitely until timeout when receiving sensor data timestamp in Navigation

Elaina-rascal, 2025-04-28:
### Operating System:

Ubuntu22.04 docker

### ROS version or commit hash:

humble **geometry2 Version**: 0.25.11

### RMW implementation (if applicable):

rmw_fastrtps_cpp 

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

## Description:
When using navigation in ROS 2, the `tf2_ros::MessageFilter` keeps blocking indefinitely until a timeout occurs while receiving the sensor data timestamp. The error encountered is as follows:

```plaintext
[controller_server-1] [INFO] [1745337846.520284079] [local_costmap.local_costmap]: Message Filter dropping message: frame 'laser_frame' at time 1745337846.120 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
```

I have traced this issue to the usage of `tf2_ros::MessageFilter`. The following code snippet from the navigation system uses `tf2_ros::MessageFilter` for sensor data:

```cpp
auto filter = std::make_shared<tf2_ros::MessageFilter<sensor_msgs::msg::PointCloud2>>(
                *sub, *tf_, global_frame_, 50,
                node->get_node_logging_interface(),
                node->get_node_clock_interface(),
                tf2::durationFromSec(transform_tolerance));
```

In addition to this, I have encountered an issue related to the `tf` buffer update thread. I believe this is the root cause of the blocking behavior.

## Related Issue:
- [ROS 2 Geometry2 Issue #438](https://github.com/ros/geometry2/issues/438)

## Steps to Reproduce:
I created a Python test script to simulate the behavior:

```python
#!/usr/bin/env python3
import rclpy
import rclpy.duration
import rclpy.executors
from rclpy.node import Node
import message_filters
import tf2_ros
from sensor_msgs.msg import LaserScan
from tf2_ros import LookupException, ConnectivityException, ExtrapolationException

class LaserScanSubscriber(Node):
    def __init__(self):
        super().__init__('laser_scan_subscriber_test')

        # 固定参数
        self.topic_name = '/MS200/scan'
        self.global_frame = 'map'
        self.transform_tolerance = 1.5  # 容忍时间（秒）

        # 创建tf2 Buffer和Listener
        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self, spin_thread=True)

        # message_filters subscriber
        self.scan_sub = self.create_subscription(
            LaserScan,
            self.topic_name,
            self.scan_callback,
            1
        )
        self.get_logger().info('LaserScan Subscriber initialized.')

    def scan_callback(self, msg: LaserScan):
        # 每次收到数据，尝试查找tf变换
        print("flag\n")
        try:
            # 打印缓冲区
            self.get_logger().info(f"Buffer size: {self.tf_buffer.get_latest_common_time(self.global_frame, msg.header.frame_id)}")
            stamp = msg.header.stamp
            stamp.sec += 1
            now = self.get_clock().now()
            trans = self.tf_buffer.lookup_transform(
                self.global_frame,
                msg.header.frame_id,
                stamp,
                timeout=rclpy.duration.Duration(seconds=self.transform_tolerance)  # 设置容忍时间
            )
            self.get_logger().info(
                f"Got transform! {msg.header.frame_id} -> {self.global_frame} at {msg.header.stamp.sec}.{msg.header.stamp.nanosec}")
        except (LookupException, ConnectivityException, ExtrapolationException) as e:
            self.get_logger().warn(
                f"Dropped message at {msg.header.stamp.sec}.{msg.header.stamp.nanosec}: {str(e)}")

def main(args=None):
    rclpy.init(args=args)
    node = LaserScanSubscriber()
    exe = rclpy.executors.MultiThreadedExecutor()
    exe.add_node(node)
    exe.spin()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

When using a `MultiThreadedExecutor` in the main function, it runs correctly.

However, with the default `SingleThreadedExecutor`, it produces the same error message.

When I tried to translate the logic to C++ and switched the executor to a `MultiThreadedExecutor` and added a `TransformListener`, it still didn't work.

### Expected behavior

The `tf2_ros::MessageFilter` should not block indefinitely, and it should handle the transform timeout properly, especially when running in a multi-threaded context or using a `TransformListener`.

### Actual behavior

When the sensor data is later than the latest timestamp in the tf buffer, the `tf2_ros::MessageFilter` should wait within the tolerance time. When a new transform arrives, it should trigger the callback.

### Additional information

## Possible Cause:
The issue could be related to the `tf` buffer update thread not functioning correctly.

wjwwood, 2025-05-13:
Thanks for the detailed issue, but I think we need more information, like what source of `/tf` data did you use to test the given python script? and can you include the entire C++ version that mentioned that had the same problem?

Also, if you could link to the place in the nav2 code that uses the `tf2_ros::MessageFilter` then it would help me to know if that code is using a transform listener in a separate thread or not.

Elaina-rascal, 2025-05-14:
I updated my Docker, and the issue disappeared. My current geometry2 version is 0.25.12.

Elaina-rascal, 2025-05-14:
## TF2 Message Filter Composable Node Issue

I wrote a composable test node using `tf2::MessageFilter`.

### Behavior

- When launching it via a launch file—regardless of whether I use `executable='component_container'` or `executable='component_container_mt'`—it works without issues.
- However, when I run it directly using:

  ```bash
  ros2 run MessageFilter_node
  ```

 a threading issue occurs, which appears to be the same error I encountered before.This seems to be related to how composable nodes behave outside of a component container.

### Files (placeholders)

- C++ Source File:
 ``` cpp
#include <chrono>
#include <memory>
#include <string>

#include "geometry_msgs/msg/transform_stamped.hpp"
#include "message_filters/subscriber.h"
#include "rclcpp/rclcpp.hpp"
#include "rclcpp_components/register_node_macro.hpp"
#include "sensor_msgs/msg/laser_scan.hpp"
#include "tf2/exceptions.h"
#include "tf2_ros/buffer.h"
#include "tf2_ros/create_timer_ros.h"
#include "tf2_ros/message_filter.h"
#include "tf2_ros/transform_listener.h"

using std::placeholders::_1;

class MessageFilter : public rclcpp::Node
{
public:
    explicit MessageFilter(const rclcpp::NodeOptions &options)
        : Node("laser_scan_subscriber_with_filter", options)
    {
        // Declare and get parameters
        topic_name_ = this->declare_parameter<std::string>("topic_name", "/laser/sd/raw");
        global_frame_ = this->declare_parameter<std::string>("global_frame", "map");
        transform_tolerance_ = this->declare_parameter<double>("transform_tolerance", 1.5);
        time_offset_ = this->declare_parameter<double>("time_offset", 0.0);
        use_tf_listener_ = this->declare_parameter<bool>("use_tf_listener", true);
        // TF2 setup
        tf_buffer_ = std::make_shared<tf2_ros::Buffer>(this->get_clock());
        auto timer_interface = std::make_shared<tf2_ros::CreateTimerROS>(
            this->get_node_base_interface(),
            this->get_node_timers_interface());
        tf_buffer_->setCreateTimerInterface(timer_interface);
        if (use_tf_listener_)
        {
            tf_listener_ = std::make_shared<tf2_ros::TransformListener>(*tf_buffer_);
        }
        // message_filters subscriber + tf2 message filter
        laser_sub_ = std::make_shared<message_filters::Subscriber<sensor_msgs::msg::LaserScan>>(
            this, topic_name_, rmw_qos_profile_sensor_data);

        tf_filter_ = std::make_shared<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>>(
            *laser_sub_,
            *tf_buffer_,
            global_frame_,
            10, // queue size
            this->get_node_logging_interface(),
            this->get_node_clock_interface());

        tf_filter_->setTolerance(rclcpp::Duration::from_seconds(transform_tolerance_));
        tf_filter_->registerCallback([this](const sensor_msgs::msg::LaserScan::ConstSharedPtr msg)
                                     { scanCallback(*msg); });

        RCLCPP_INFO(this->get_logger(), "LaserScan Node with MessageFilter Initialized.");
    }

private:
    void scanCallback(const sensor_msgs::msg::LaserScan msg)
    {

        RCLCPP_INFO(this->get_logger(), "Got transform (via filter): %s -> %s at time %u.%u",
                    msg.header.frame_id.c_str(),
                    global_frame_.c_str(),
                    msg.header.stamp.sec,
                    msg.header.stamp.nanosec);
    }

    std::string topic_name_;
    std::string global_frame_;
    double transform_tolerance_;
    double time_offset_;
    bool use_tf_listener_;
    std::shared_ptr<tf2_ros::Buffer> tf_buffer_;
    std::shared_ptr<tf2_ros::TransformListener> tf_listener_;

    std::shared_ptr<message_filters::Subscriber<sensor_msgs::msg::LaserScan>> laser_sub_;
    std::shared_ptr<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>> tf_filter_;
};

RCLCPP_COMPONENTS_REGISTER_NODE(MessageFilter)

```

- Launch File: 
```python
from launch import LaunchDescription
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    return LaunchDescription([
        ComposableNodeContainer(
            name='laser_scan_container',
            namespace='',
            package='rclcpp_components',
            # executable='component_container_mt',
            executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='rc_navigation',
                    plugin='MessageFilter',
                    name='laser_scan_node',
                    parameters=[{
                        'topic_name': '/laser/sd/raw',
                        'global_frame': 'map',
                        'transform_tolerance': 0.3
                            
                    }]
                )
            ],
            output='screen',
        )
    ])

```

### Related Reference

The original C++ source that had a similar issue can be found here:

https://github.com/ros-navigation/navigation2/blob/humble/nav2_costmap_2d/plugins/obstacle_layer.cpp

The relevant MessageFilter-related code starts at line 225.