ros2/geometry2 #798: No authority available

roncapat, 2025-06-18:
### Description

`tf2_monitor` produces the output `published by <no authority detectable`.

There is this pending TODO (@tfoote any clue on how to proceed?).
https://github.com/ros2/geometry2/blob/141e3cb732e5c4dba744abac381c8317ffb286f0/tf2_ros/src/transform_listener.cpp#L98-L99

### Motivation

Solve the aforementioned TODO or understand current limitation in ROS 2 framework that prevents this issue.

tfoote, 2025-06-18:
You look to have found the best references. We had this ability in the past, but it's kind of counter to the philosophy of anonymous publish subscribe.  But if we can get it such as from the rosbag improvements it would be helpful.

roncapat, 2025-06-19:
Could you clarify the last sentence? I don't get the rosbag reference, even if I read the linked rosbag2 PR

tfoote, 2025-06-20:
Sorry I hadn't realized that the referenced PR had captured the authority already, I was referring to rosbag having solved that problem, but clearly sloretz did here too.. 

Once the authority is inserted into the buffer via setTransform we used to have a method which would dump out the whole dot graph in the buffer core. https://github.com/ros/geometry2/blob/8fef92a6b56a26e2b5be58d4d5c84dc4de5ee696/tf2/src/buffer_core.cpp#L1478
https://github.com/ros/geometry2/blob/noetic-devel/tf2/src/buffer_core.cpp
Reviving that would avoid the need for tf2_monitor to reimplement/duplicate that logic. 

Landing #545 in itself would improve the debug info from TF if people turn it on. It should populate a lot of messages that use the authority info in https://github.com/ros/geometry2/blob/noetic-devel/tf2/src/buffer_core.cpp

CursedRock17, 2025-08-27:
Hey all, any status update on the related PR? If not I would always be down to throw a new one in.